d-a-v/W5500lwIP #1: ESP32 support ?

psykokwak-com, 2018-06-05:
Hi,
Do you plan to add ESP32 support ?
I tried to build it but it failed.

```
Jyce@XPS15 MINGW32 ~/esp/arduino
$ make -j4 flash
CXX build/main/w5500.o
In file included from C:/msys32/home/Jyce/esp/esp-idf/components/esp32/include/xtensa/xtruntime.h:30:0,
                 from C:/msys32/home/Jyce/esp/esp-idf/components/freertos/include/freertos/portmacro.h:81,
                 from C:/msys32/home/Jyce/esp/esp-idf/components/freertos/include/freertos/portable.h:94,
                 from C:/msys32/home/Jyce/esp/esp-idf/components/freertos/include/freertos/FreeRTOS.h:105,
                 from C:/msys32/home/Jyce/esp/arduino/components/arduino/cores/esp32/Arduino.h:32,
                 from C:/msys32/home/Jyce/esp/arduino/main/w5500.h:37,
                 from C:/msys32/home/Jyce/esp/arduino/main/w5500.cpp:33:
C:/msys32/home/Jyce/esp/esp-idf/components/esp32/include/xtensa/config/specreg.h:102:13: error: expected identifier before numeric constant
 #define MR  32
             ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:300:9: note: in expansion of macro 'MR'
         MR = 0x0000,        ///< Mode Register address (R/W)
         ^
C:/msys32/home/Jyce/esp/esp-idf/components/esp32/include/xtensa/config/specreg.h:102:13: error: expected '}' before numeric constant
 #define MR  32
             ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:300:9: note: in expansion of macro 'MR'
         MR = 0x0000,        ///< Mode Register address (R/W)
         ^
C:/msys32/home/Jyce/esp/esp-idf/components/esp32/include/xtensa/config/specreg.h:102:13: error: expected unqualified-id before numeric constant
 #define MR  32
             ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:300:9: note: in expansion of macro 'MR'
         MR = 0x0000,        ///< Mode Register address (R/W)
         ^
In file included from C:/msys32/home/Jyce/esp/arduino/main/w5500.cpp:33:0:
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'void setMR(uint8_t)':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:459:23: error: 'BlockSelectCReg' was not declared in this scope
         wizchip_write(BlockSelectCReg, MR, mode);
                       ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:459:48: error: 'wizchip_write' was not declared in this scope
         wizchip_write(BlockSelectCReg, MR, mode);
                                                ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'uint8_t getMR()':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:468:29: error: 'BlockSelectCReg' was not declared in this scope
         return wizchip_read(BlockSelectCReg, MR);
                             ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:468:48: error: 'wizchip_read' was not declared in this scope
         return wizchip_read(BlockSelectCReg, MR);
                                                ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'void setSHAR(const uint8_t*)':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:477:27: error: 'BlockSelectCReg' was not declared in this scope
         wizchip_write_buf(BlockSelectCReg, SHAR, macaddr, 6);
                           ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:477:44: error: 'SHAR' was not declared in this scope
         wizchip_write_buf(BlockSelectCReg, SHAR, macaddr, 6);
                                            ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:477:60: error: 'wizchip_write_buf' was not declared in this scope
         wizchip_write_buf(BlockSelectCReg, SHAR, macaddr, 6);
                                                            ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'void getSHAR(uint8_t*)':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:486:26: error: 'BlockSelectCReg' was not declared in this scope
         wizchip_read_buf(BlockSelectCReg, SHAR, macaddr, 6);
                          ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:486:43: error: 'SHAR' was not declared in this scope
         wizchip_read_buf(BlockSelectCReg, SHAR, macaddr, 6);
                                           ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:486:59: error: 'wizchip_read_buf' was not declared in this scope
         wizchip_read_buf(BlockSelectCReg, SHAR, macaddr, 6);
                                                           ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'void setIR(uint8_t)':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:495:23: error: 'BlockSelectCReg' was not declared in this scope
[... 180 lines cut ...]
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'uint16_t getSn_RX_RD()':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:706:34: error: 'BlockSelectSReg' was not declared in this scope
         return wizchip_read_word(BlockSelectSReg, Sn_RX_RD);
                                  ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:706:59: error: 'wizchip_read_word' was not declared in this scope
         return wizchip_read_word(BlockSelectSReg, Sn_RX_RD);
                                                           ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: In function 'uint16_t getSn_RX_WR()':
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:714:34: error: 'BlockSelectSReg' was not declared in this scope
         return wizchip_read_word(BlockSelectSReg, Sn_RX_WR);
                                  ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:714:59: error: 'wizchip_read_word' was not declared in this scope
         return wizchip_read_word(BlockSelectSReg, Sn_RX_WR);
                                                           ^
C:/msys32/home/Jyce/esp/arduino/main/w5500.h: At global scope:
C:/msys32/home/Jyce/esp/arduino/main/w5500.h:716:1: error: expected declaration before '}' token
 };
 ^
make[1]: *** [/home/Jyce/esp/esp-idf/make/component_wrapper.mk:285: w5500.o] Error 1
make: *** [C:/msys32/home/Jyce/esp/esp-idf/make/project.mk:467: component-main-build] Error 2
make: *** Attente des tâches non terminées....

Jyce@XPS15 MINGW32 ~/esp/arduino

```

psykokwak-com, 2018-06-05:
Thank for your answer.
I fixed all the compilation errors but the ESP32 crashes when it call begin()

```
C:/msys32/home/Jyce/esp/esp-idf/components/freertos/queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x400869b5 on core 1
0x400869b5: xQueueGenericReceive at C:/msys32/home/Jyce/esp/esp-idf/components/freertos/queue.c:1995

Backtrace: 0x4008815c:0x3ffbb290 0x40088337:0x3ffbb2b0 0x400869b5:0x3ffbb2d0 0x4010da72:0x3ffbb310 0x4010db2e:0x3ffbb330 0x401049b0:0x
3ffbb350 0x40109167:0x3ffbb370 0x40109205:0x3ffbb390 0x40109b5d:0x3ffbb3b0 0x400ecfa1:0x3ffbb3d0 0x400ecfe5:0x3ffbb410 0x400e71e6:0x3f
fbb440 0x400e9e7f:0x3ffbb480 0x400e422f:0x3ffbb4b0
0x4008815c: invoke_abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:648
0x40088337: abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:648
0x400869b5: xQueueGenericReceive at C:/msys32/home/Jyce/esp/esp-idf/components/freertos/queue.c:1995
0x4010da72: sys_mutex_lock at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/port/freertos/sys_arch.c:468
0x4010db2e: sys_arch_protect at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/port/freertos/sys_arch.c:486
0x401049b0: pbuf_free at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/pbuf.c:883
0x40109167: dhcp_delete_msg at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/ipv4/dhcp.c:568
0x40109205: dhcp_discover at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/ipv4/dhcp.c:568
0x40109b5d: dhcp_start at C:/msys32/home/Jyce/esp/esp-idf/components/lwip/core/ipv4/dhcp.c:842
0x400ecfa1: Wiznet5500lwIP::start_with_dhclient() at C:/msys32/home/Jyce/esp/arduino/main/w5500-lwIP.cpp:78
0x400ecfe5: Wiznet5500lwIP::begin(SPIClass&, unsigned char const*, unsigned short) at C:/msys32/home/Jyce/esp/arduino/main/w5500-lwIP.cpp:27
0x400e71e6: networkETHInitialize() at C:/msys32/home/Jyce/esp/arduino/main/SentinelSimpleFirmwareMain.cpp:1485
0x400e9e7f: setup() at C:/msys32/home/Jyce/esp/arduino/main/SentinelSimpleFirmwareMain.cpp:1485
0x400e422f: loopTask(void*) at C:/msys32/home/Jyce/esp/arduino/components/arduino/cores/esp32/main.cpp:15

Rebooting...
```

d-a-v, 2018-06-05:
You may need to properly initialize the internal network on esp32 before trying to use this external interface.
I can't help further, check the source before the abort() call
```
0x40088337: abort at C:/msys32/home/Jyce/esp/esp-idf/components/esp32/panic.c:648
0x400869b5: xQueueGenericReceive at C:/msys32/home/Jyce/esp/esp-idf/components/freertos/queue.c:1995
```

psykokwak-com, 2018-06-12:
I improved the ESP32 support. See https://github.com/d-a-v/W5500lwIP/pull/2

johnnytolengo, 2018-11-11:
> `Thank for your answer.
I fixed all the compilation errors but the ESP32 crashes when it call begin()`

How do you solve the MR,etc definitions?

johnnytolengo, 2018-11-11:
I got the same crash:
`[I][w5500-lwIP.cpp:121] begin(): NO_SYS=0
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40084b75 on core 1
`
Could you help me how to fix it?
Thank you.

d-a-v, 2018-11-12:
@psykokwak-com @johnnytolengo For now I can't help with the errors you're seeing.

@me-no-dev,
Have you had a chance to try Wiznet5500 SPI ethernet chips on esp32/arduino ?

psykokwak-com, 2018-11-12:
@johnnytolengo, Try this fork : https://github.com/psykokwak-com/W5500lwIP
@me-no-dev, can you give us some input to help ?

d-a-v, 2018-11-12:
@johnnytolengo You may try with @psykokwak-com's branch
https://github.com/psykokwak-com/W5500lwIP/tree/esp32

johnnytolengo, 2018-11-12:
I already tried @psykokwak-com's branch
https://github.com/psykokwak-com/W5500lwIP/tree/esp32
**With some modifications it compiled:**
`w5500-lwIP.cpp
-- boolean Wiznet5500lwIP::begin (SPIparam(SPIClass& spi,) const uint8_t* macAddress, uint16_t mtu)
++ boolean Wiznet5500lwIP::begin(const uint8_t* macAddress, uint16_t mtu)
w5500-lwIP.h

-- boolean begin (SPIparam(SPIClass& spi,) const uint8_t *macAddress = NULL, uint16_t mtu = 1500); MANU
++ boolean begin (const uint8_t *macAddress = NULL, uint16_t mtu = 1500);`

**Then using:**
Wiznet5500lwIP ether(CSPIN)
uint8_t mac_addr[6] = { 0xDE, 0xAD, 0xBE, 0xEF, 0xFE, 0xE };
call ether.begin(mac_addr) in arduino's setup()
call ether.loop() in arduino's loop()

**Crashes:**

`[I][w5500-lwIP.cpp:121] begin(): NO_SYS=0
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40084b75 on core 1

Backtrace: 0x400878f4:0x3ffcc050 0x40087acf:0x3ffcc070 0x40084b75:0x3ffcc090 0x40108e2a:0x3ffcc0d0 0x40109066:0x3ffcc0f0 0x400fd104:0x3ffcc110 0x40102e8f:0x3ffcc130 0x40102f2d:0x3ffcc150 0x40103885:0x3ffcc170 0x400d30ae:0x3ffcc190 0x400d30f8:0x3ffcc1d0 0x400d251b:0x3ffcc210 0x400d2bbe:0x3ffcc240 0x400d2cdf:0x3ffcc280 0x4010c8cf:0x3ffcc2f0
`

psykokwak-com, 2018-11-12:
@johnnytolengo, did you start the wifi before ?
The W5500 lwip driver does not start the TCPIP stack. So you have to do it or start the wifi before (that starts the TCPIP stack for you).

johnnytolengo, 2018-11-12:
Ohh I didn't know that the wifi must be "ON". 
Now it is on and just using  ether.begin(mac_addr)  the W5500 is now on and got IP from the DHCP also reply to ping. But I still have this error log:

[image]

Otherwise using : ether.begin(mac_addr) and ether.loop() I have this error log:

[image]

In this case the W5500 is not getting IP from the DHCP.

johnnytolengo, 2018-11-12:
@psykokwak-com I did the test again and I it is working without ether.loop().  

 I be glad to know how can I set the IP address for the module :)

Thank you

psykokwak-com, 2018-11-12:
@johnnytolengo, Yes, this implementation create its own task for loop.
Right now there is no simple way to force an IP address but it should not be hard to implement.
Be careful, its still unstable especialy if you use the wifi in the same time.

johnnytolengo, 2018-11-12:
Could you give me a tip for what I have to do to add "set ip address" feature?
I am a noob in the lwIP world, I really don't know how it works and maybe with some tips I will able start and develop it.
Thank you.

d-a-v, 2018-11-12:
@psykokwak-com You mention unstability in #2.
Can you please elaborate ?

@johnnytolengo Static address is doable.
I will try at dome point with something like this:
```
err_t Wiznet5500lwIP::start_with_static_address (uint32_t ip32, uint32_t mask32, uint32_t gw32)
{
    ip4_addr_t ip, mask, gw;

    ip4_addr_set_u32(&ip, ip32);
    ip4_addr_set_u32(&mask, mask32);
    ip4_addr_set_u32(&gw, gw32);

    _netif.hwaddr_len = sizeof _mac_address;
    memcpy(_netif.hwaddr, _mac_address, sizeof _mac_address);

    if (!netif_add(&_netif, &ip, &mask, &gw, this, netif_init_s, ethernet_input))
        return ERR_IF;

    _netif.flags |= NETIF_FLAG_UP;
    _netif.flags |= NETIF_UP;

    return ERR_OK;
}
```

johnnytolengo, 2018-11-12:
Ok I will work on that. 
Do you have also any tips how to make a EthClient  ?

d-a-v, 2018-11-12:
Like arduino's `WiFiClient` ?
Just use `WiFiClient`, it is supposed to work that way.
Output interface is chosen by lwIP according to destination, mask and gateway address matching.

johnnytolengo, 2018-11-12:
ok, but what happen if Wifi and Ethernet are in different states?
Eg.
1) Wifi in WiFi.mode(WIFI_OFF)  and Ethernet connected via DHCP.
2) Wifi and Ethernet in a different networks with different gateways eg.:

     - Wifi 192.168.1.100 (local without access to internet) and 
      - Ethernet 192.168.22.100 (fully connected to interenet) 

The thing is how to manage the default gateway?

Thanks

johnnytolengo, 2018-11-16:
After a week finally I have the static ip address working, BUT... the gateway is not working (seems to be the default route) inside the local network everything work.
That connection (google.com) works with WIFI but **not with Ethernet**:

**Any idea how to set the default gateway for both networks or at least how to make work each gateway for each interface?**

`WiFiClient wifiClient;
char g_server[] = "172.217.23.206"; //or google.com
  
  if ( wifiClient.connect(g_server, 80) ) {
      Serial.println("connected to google");

  }`

**Here the working static IP:**
`err_t Wiznet5500lwIP::start_with_static_address ()
{
    ip4_addr_t ip, mask, gw;
    uint32_t new_ip[] = { 0xC0, 0xA8, 0x01, 0x96};
    
    //that did the trick  "LWIP_DHCP" for static ip address:) stops the DHCP
    #undef LWIP_DHCP
    #define LWIP_DHCP 0

    IP4_ADDR( &ip, 192,168,1,50 );
    IP4_ADDR( &mask,255,255,255,0 );
    IP4_ADDR( &gw, 192,168,1,1 );

    _netif.hwaddr_len = sizeof _mac_address;
    memcpy(_netif.hwaddr, _mac_address, sizeof _mac_address);

    _callbackmsg = tcpip_callbackmsg_new(_tcpcallback, this);

    if (!netif_add(&_netif, &ip, &mask, &gw, this, netif_init_s, ethernet_input)){
    
        netif_set_default(&_netif);
        return ERR_IF;
    }
    
    ESP_LOGW("w5500-lwip", "Current ip: %d:%d:%d:%d", localIP()[0],localIP()[1],localIP()[2],localIP()[3] );
    ESP_LOGW("w5500-lwip", "Current mask: %d:%d:%d:%d", subnetMask()[0],subnetMask()[1],subnetMask()[2],subnetMask()[3] );
    ESP_LOGW("w5500-lwip", "Current gateway: %d:%d:%d:%d", gatewayIP()[0],gatewayIP()[1],gatewayIP()[2],gatewayIP()[3] );

    _netif.flags |= NETIF_FLAG_UP;

    return ERR_OK;
}`

johnnytolengo, 2018-11-18:
Finally works! at least the basics :)
Here is my code:
https://github.com/johnnytolengo/w5500lwip

d-a-v, 2018-11-19:
Congrats!

psykokwak-com, 2018-11-19:
@johnnytolengo , Why did you remove all SPIparam(x...) and SPI transaction functions ? This is not complient with the SPI standard.

johnnytolengo, 2018-11-19:
SPI is handled by w5500.cpp lib. It takes the default spi port.

[image]

psykokwak-com, 2018-11-19:
SPI is a multi devices bus. So, its important to do not initialize the SPI library inside the device lib and use the right SPI implementation with beginTransaction() and endTransaction().

johnnytolengo, 2018-11-19:
@johnnytolengo 's repository it is only for ESP32. About the SPI will be fine to add an option to choose the bus you can use, I am agree.
Right now my repository has a very raw first version.

d-a-v, 2018-11-19:
> Right now my repository has a very raw first version.

Pretty much like my repository is for esp8266. Not everything is working yet.