d-a-v/W5500lwIP #5: Add example to the repo

Pimmetje, 2019-05-10:
It would be nice to add the example given here https://github.com/d-a-v/W5500lwIP/issues/3#issuecomment-431273815 as a example to the repository. 

It could save some people some searching for the correct pin connection. 

Thanks for making this. Ill test it later today.

d-a-v, 2019-05-10:
Thanks for your interest. Current master will not work, use commit 81fd79b10955a436d111b984078a97d64801eb8f instead. Explanations:

This driver is about to be integrated in the esp8266 arduino core.
esp8266/arduino core-2.5.1 is about to be released and right after that,
per https://github.com/d-a-v/W5500lwIP/commit/b468cc893e655752cb5876dabb310950fb11751c and this awaiting PR https://github.com/esp8266/Arduino/pull/6039 which will hopefully be integrated,
utilization of this driver will be much simpler and will fix use-cases like the one exposed by @borisneubert in https://github.com/d-a-v/W5500lwIP/issues/3#issuecomment-435566310

I am also currently working on a driver compatible with W5100 (#4)

```
// D0 CS SDCARD
// D8 CS ethernet
// D3 MOS
// D2 INPUT WATER
// D4 1-WIRE temp

#include <ESP8266WiFi.h>

#include <w5500-lwIP.h>

Wiznet5500lwIP eth(D0);

void setup ()
{
  // setup code
  // ...

  eth.setDefault(); // use ethernet for default route
  eth.begin(); // default mtu & mac address
}

void loop ()
{
  // do whatever needed, including
  while (true)
  {
    // do stuff
    yield();
  }
}
```

Pimmetje, 2019-05-10:
Thanks for the fast response

I tried several versions of ESP the error i get is the same. It seems to me a issue with a include or something.
```
C:\Users\Pim\Documents\Arduino\libraries\W5500lwIP\w5500-lwIP.cpp: In member function 'boolean Wiznet5500lwIP::begin(const uint8_t*, uint16_t)':

C:\Users\Pim\Documents\Arduino\libraries\W5500lwIP\w5500-lwIP.cpp:52:86: error: 'schedule_function_us' was not declared in this scope

     else if (!schedule_function_us([&]() { this->handlePackets(); return true; }, 100))

                                                                                      ^

In file included from C:\Users\Pim\Documents\Arduino\libraries\W5500lwIP\w5500-lwIP.cpp:12:0:

C:\Users\Pim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/FunctionalInterrupt.h: At global scope:

C:\Users\Pim\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.4.2\cores\esp8266/FunctionalInterrupt.h:32:28: warning: 'scheduledInterrupts' defined but not used [-Wunused-variable]

 static ScheduledFunctions* scheduledInterrupts;
```

d-a-v, 2019-05-10:
> I tried several versions of ESP the error i get is the same.

As I told, 
> > Current master will not work, use commit 81fd79b instead.

Or, try the PR https://github.com/esp8266/Arduino/pull/6039

borisneubert, 2019-05-10:
Thank you d-a-v, still watching. Will try soon.

d-a-v, 2019-05-23:
Current state of this driver will now work with latest git version esp8266 arduino core.
(I sufferred to get this PR in)

borisneubert, 2019-05-30:
Tried this example:

```
#include <Arduino.h>
#include "ESP8266WiFi.h"
#include <W5500lwIP.h>

// Arduino Pin 4 = Wemos Pin D2
#define CSPIN 4

Wiznet5500lwIP ether;

byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

void setup() {

  Serial.begin(115200);
  Serial.println("");
  Serial.println("start");

  ether.setDefault();
  //int present = ether.begin(mac);
  int present = ether.begin();
  Serial.println("present= " + String(present, HEX));

  while (!ether.connected()) {
    Serial.print(".");
    delay(1000);
  }
}

void loop() {  }

```

with latest [Arduino core](https://github.com/esp8266/Arduino).

Still get this error:

```
In file included from .pio/libdeps/d1_mini/W5500lwIP/src/W5500lwIP.h:5:0,
                 from src/main.cpp:3:
.pio/libdeps/d1_mini/W5500lwIP/src/utility/lwIPeth.h: In instantiation of 'boolean LwipEthernet<RawEthernet>::begin(const uint8_t*, uint16_t) [with RawEthernet = Wiznet5500; boolean = bool; uint8_t = unsigned char; uint16_t = short unsigned int]':
src/main.cpp:20:29:   required from here
.pio/libdeps/d1_mini/W5500lwIP/src/utility/lwIPeth.h:127:86: error: 'schedule_function_us' was not declared in this scope
     else if (!schedule_function_us([&]() { this->handlePackets(); return true; }, 100))
                                                                                      ^
```

d-a-v, 2019-05-31:
Sorry you need to use git version.
The api for Ethernet is still not stable and under heavy discussion.
It should work anyway and I'll update when needed until I can integrate 
everything into the arduino core.

borisneubert, 2019-05-31:
Thank you for your quick reply!

I actually use the version of W5500lwIP from this git repository. 

We just wait until things have stabilized and continue to watch this repo.

emelianov, 2019-06-18:
For some reason at least with latest Arduino core (commit 9f03bbb8c38adf2f6dc9fb95e304b51f20a72f92
from Jun 14 14:58:29 2019) handlePackets() never called with delay() or yield() in setup section. Have to add run_scheduled_functions() as workaround.
```c
#include <SPI.h>
#include <ESP8266WiFi.h>

#include <W5500lwIP.h>
#include <Schedule.h>

Wiznet5500lwIP eth(SPI, D4);
byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

void setup ()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_OFF);
  SPI.begin();
  eth.setDefault(); // use ethernet for default route
  int present = eth.begin(mac);
  Serial.println("present= " + String(present, HEX));

  while (!eth.connected()) {
    Serial.print(".");
    delay(1000);
    run_scheduled_functions();
  }
  Serial.println(eth.localIP());
}

void loop ()
{
    yield();
}
```

d-a-v, 2019-09-19:
@borisneubert 
I have so far been using one of the examples you pointed me to to test this driver.
(with mqtt)

```

#include <Arduino.h>

//....................................................
// WiFi
#include "ESP8266WiFi.h"
const char *ssid = STASSID;
const char *password = STAPSK;

//....................................................
#define CSPIN 16 // GPIO16

//#include <W5500lwIP.h>
//Wiznet5500lwIP eth(SPI, CSPIN);

#include <W5100lwIP.h>
Wiznet5100lwIP eth(SPI, CSPIN);

int present = 0;

#include <osapi.h>
LOCAL os_timer_t eth_timer;
int led = 0;

//....................................................
// asynchronous web server
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
AsyncWebServer server(80);

//....................................................
// ntp client
#include <NTPClient.h>
#include <WiFiUdp.h>

#define NTPSERVER "91.202.42.83"
//#define NTPSERVER "0.ubuntu.pool.ntp.org"
//#define NTPSERVER "192.53.103.108"
WiFiUDP ntpUDP;
NTPClient ntpClient(ntpUDP, NTPSERVER, 3600, 20000);
unsigned long long t = 0, t0 = 0;

//....................................................
// mqtt client
#include <PubSubClient.h>
WiFiClient espClient;
PubSubClient client(espClient);
const char* mqtt_server = "10.0.1.254";
long lastMsg = 0;
char msg[50];
int value = 0;
String mqttMsg;

// mqtt callback
void callback(char* topic, byte* payload, unsigned int length) {
  Serial.println("MQTT incomming subcribe: ");
  mqttMsg = topic;
  mqttMsg += " - ";
  for (int i = 0; i < length; i++) {
    mqttMsg += (char)payload[i];
  }
  Serial.println(mqttMsg);
}
// mqtt reconnect
void reconnect() {
[... 146 lines cut ...]

    t = ntpClient.getEpochTime();
    if (t != t0) {
      if (!client.connected()) reconnect();

      Serial.println(ntpClient.getFormattedTime());
      Serial.print("wifi ip address: ");
      Serial.println(WiFi.localIP());
      Serial.print("ethernet ip address: ");
      Serial.println(eth.localIP());

      client.publish("outTopic", ntpClient.getFormattedTime().c_str());
      t0 = t;

      uint32_t mfree;
      uint16_t mmax;
      uint8_t frag;
      ESP.getHeapStats(&mfree, &mmax, &frag);
      Serial.printf("%d %d %d\n", mfree, mmax, frag);

    }
    yield();
  }
}
```

Pfannex, 2019-10-20:
Moin d-a-v,

I tried to run your latest example. 
I use your latest W5500lwIP commit an the actual stable Arduino core version 2.52.

Compiling the demo sketch results the following errors:

> In file included from 
\Arduino\libraries\W5500lwIP/W5100lwIP.h:5:0,
from W5500_dav_NEW.ino:15:
Documents\Arduino\libraries\W5500lwIP/utility/lwIPeth.h: In instantiation of 'boolean LwipEthernet<RawEthernet>::begin(const uint8_t*, uint16_t) [with RawEthernet = Wiznet5100; boolean = bool; uint8_t = unsigned char; uint16_t = short unsigned int]':
>
>W5500_dav_NEW.ino:146:23:   required from here
>
>Arduino\libraries\W5500lwIP/utility/lwIPeth.h:133:96: error: 'schedule_recurrent_function_us' was not declared in this scope
>
>else if (!schedule_recurrent_function_us([&]() { this->handlePackets(); return true; }, 100))

Witch core-version do you use?

d-a-v, 2019-10-20:
Latest git is required !

mynameisbill2, 2019-10-28:
> Moin d-a-v,
> 
> I tried to run your latest example.
> I use your latest W5500lwIP commit an the actual stable Arduino core version 2.52.
> 
> Compiling the demo sketch results the following errors:
> 
> > In file included from
> > \Arduino\libraries\W5500lwIP/W5100lwIP.h:5:0,
> > from W5500_dav_NEW.ino:15:
> > Documents\Arduino\libraries\W5500lwIP/utility/lwIPeth.h: In instantiation of 'boolean LwipEthernet::begin(const uint8_t*, uint16_t) [with RawEthernet = Wiznet5100; boolean = bool; uint8_t = unsigned char; uint16_t = short unsigned int]':
> > W5500_dav_NEW.ino:146:23:   required from here
> > Arduino\libraries\W5500lwIP/utility/lwIPeth.h:133:96: error: 'schedule_recurrent_function_us' was not declared in this scope
> > else if (!schedule_recurrent_function_us(& { this->handlePackets(); return true; }, 100))
> 
> Witch core-version do you use?

Hi Pfannex,
  Have you resolved this problem? I met the same problem.

d-a-v, 2019-10-28:
core 2.5.2 will not work.
Git version of the core is necessary.
Alternatively, you can use snapshots of the core (https://d-a-v.github.com)

d-a-v, 2019-10-28:
Ethernet PR on arduino core launched. 
Merge won't happen before next core release https://github.com/esp8266/Arduino/pull/6680
But I can keep alternative installable snapshots like in the post above which include the ethernet PR.

Pfannex, 2019-10-29:
Moin David,

with latest core commit dd73a18 WiFi and also ETH seems to run perfect including:
- async Web Server
- NTP-Client
- MQTT-Client

The Webserver is supported on WiFi-IP and also on ETH-IP.
Both runs continuously!

In the next step we will test the lwIP function in our [OmniESP-framework](https://github.com/Pfannex/OmniESP/wiki)...

good job!

latest test sketch
 ```c++
#include <Arduino.h>

//....................................................
// WiFi
#include "ESP8266WiFi.h"
const char *ssid = "Pf@nne-NET";
const char *password = "Pf@nneNETwlan_ACCESS";

//....................................................
#define CSPIN 4 // GPIO4

#include <W5500lwIP.h>
Wiznet5500lwIP eth(SPI, CSPIN);

//#include <W5100lwIP.h>
//Wiznet5100lwIP eth(SPI, CSPIN);

int present = 0;

#include <osapi.h>
LOCAL os_timer_t eth_timer;
int led = 0;

//....................................................
// asynchronous web server
#include <ESPAsyncTCP.h>
#include <ESPAsyncWebServer.h>
AsyncWebServer server(80);

//....................................................
// ntp client
#include <NTPClient.h>
#include <WiFiUdp.h>

#define NTPSERVER "91.202.42.83"
//#define NTPSERVER "0.ubuntu.pool.ntp.org"
//#define NTPSERVER "192.53.103.108"
WiFiUDP ntpUDP;
NTPClient ntpClient(ntpUDP, NTPSERVER, 3600, 20000);
unsigned long long t = 0, t0 = 0;

//....................................................
// mqtt client
#include <PubSubClient.h>
WiFiClient espClient;
PubSubClient client(espClient);
const char* mqtt_server = "192.168.1.10";
long lastMsg = 0;
char msg[50];
int value = 0;
String mqttMsg;

// mqtt callback
void callback(char* topic, byte* payload, unsigned int length) {
[... 159 lines cut ...]
    t = ntpClient.getEpochTime();
    if (t != t0) {
      if (!client.connected()) reconnect();

      Serial.println(ntpClient.getFormattedTime());
      Serial.print("wifi ip address: ");
      Serial.println(WiFi.localIP());
      Serial.print("ethernet ip address: ");
      Serial.println(eth.localIP());
      //Serial.println(eth.begin());

      client.publish("outTopic", ntpClient.getFormattedTime().c_str());
      t0 = t;

      uint32_t mfree;
      uint16_t mmax;
      uint8_t frag;
      ESP.getHeapStats(&mfree, &mmax, &frag);
      Serial.printf("%d %d %d\n", mfree, mmax, frag);

    }
    yield();
  }
}
```

d-a-v, 2019-10-29:
Thanks for testing, for the report !

Pfannex, 2019-12-19:
Hi David,

right now we implement your lwIP into our framework. 
It works pretty!

Every time the state of the ETH-connection changes we also want to change the default route.

Therefor you implemented:

```c++
template <class RawEthernet>
void LwipEthernet<RawEthernet>::setDefault ()
{
    Serial.println("set ETH to default, revise if not connected!");
    _default = true;
    if (connected())
        netif_set_default(&_netif);
}

#endif // _LWIPETH_H
``` 
How to make it switchable?

like
```c++
void LwipEthernet<RawEthernet>::setDefault (boolean state)
{
    _default = state;
     netif_set_default(&_netif);
}
``` 

or like

```c++
void LwipEthernet<RawEthernet>::setDefault ()
{
    if (connected()){
      _default = true;
    }else{
      _default = false;
    }
     netif_set_default(&_netif);
}
```

d-a-v, 2019-12-20:
I think it would work: ([why](http://git.savannah.nongnu.org/cgit/lwip.git/tree/src/core/netif.c?h=STABLE-2_1_2_RELEASE#n822))
```
void LwipEthernet<RawEthernet>::setDefault (boolean enable)
{
     netif_set_default(enable? &_netif: nullptr);
}
```