d-a-v/W5500lwIP #9: Cannot get this to work on a W5500 lite module

paultbarrett, 2019-08-13:
For the life of me I cannot get this library to work.

**Hardware:** 
Wemos NodeMCU 3
W5500 Lite Module

Have tested the hardware with ESP8266 Board Package 2.4.2 and Ethernet 2.0.0 library so can confirm all is working. Example sketch DhcpAddressPrinter working

Used both D8 (With transistor) and D2, both work in non W5500lwIP setup

** Testing so far **

Have installed the latest esp8266 core from git (2.6dev) along with latest git of this library.
Used various example sketches found within this issues section with no luck.

After sketch is loaded it never actually connects. Just sits within the !eth.connected loop

`#include <SPI.h>
#include <ESP8266WiFi.h>

#include <W5500lwIP.h>
#include <Schedule.h>

Wiznet5500lwIP eth(SPI, D0);
byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

void setup ()
{
  Serial.begin(115200);

  int present = eth.begin(mac);
  Serial.println("present= " + String(present, HEX));

  WiFi.mode(WIFI_OFF);

  while (!eth.connected()) {
    Serial.print(".");
    delay(1000);
    run_scheduled_functions();
  }
  Serial.println(eth.localIP());
  eth.setDefault(); // use ethernet for default route
}

void loop ()
{
    yield();
}`

paultbarrett, 2019-08-13:
Should also point out that int present = eth.begin(mac) results in 0 so it is not initialising correctly

d-a-v, 2019-08-15:
I need to take the Ethernet 2.0.0's W5100 driver and extract from it the ethernet part only (remove the IP/UDP/TCP part) so we can have W5100 W5500 and your W5500 lite working with this driver for lwIP.
I need: time :]

paultbarrett, 2019-08-16:
The W5500 lite still uses the standard W5500 chip so shouldn't be any different. Is it just that the library doesn't work anymore with any W5500s ?

If you can tell me the version of esp library and matching W5500lwIP library then I am happy to downgrade

d-a-v, 2019-08-16:
`eth.begin()` does not depend on arduino core. The current driver only works with W5500, so it should work. I'll try on my tonight (12 hours from now) with my hardware and confirm it is still working.

d-a-v, 2019-08-17:
I just tried again and it works.
Note that you need to initialize SPI before calling `.begin()`.
(SPI is not initialized from inside this ethernet driver, not like ethernet 2.0.0 which does the init)

paultbarrett, 2019-08-17:
That was it.

I added SPI.begin(); before the eth.begin and it is now working.

Thanks for the help.

closed 2019-08-17 by paultbarrett