d-a-v/W5500lwIP #10: The Value of @ref Sn_SR always equals 0

mynameisbill2, 2019-11-12:
I'm using the W5500 like this,
[image]
My notes on the connections I’m using are:-
ESP8266    W5500
GPIO5        RST
GPIO14      SCLK
GPIO12      MISO
GPIO13      MOSI
GPIO4        SCS
3.3v            3.3v
GND           GND

the code like this:

`#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <W5500lwIP.h>
#include <SPI.h>

// Arduino Pin 4 = Wemos Pin D2
#define CSPIN 4

Wiznet5500lwIP ether(SPI, CSPIN);  

byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

void setup() {

  Serial.begin(115200);
  Serial.println("");
  Serial.println("start");

  ether.setDefault();
  //bool present = ether.begin(mac);
  bool present = ether.begin();
  // 输出结果0=false, 1=true
  Serial.printf("present= %d\r\n", present);

  while (!ether.connected()) {
    Serial.print(".");
    delay(1000);
  }

  Serial.println("connected");
}

void loop() {  }`

but i show the @ref Sn_SR always equals 0, in code

`boolean Wiznet5500::begin(const uint8_t *mac_address)
{
    memcpy(_mac_address, mac_address, 6);

    pinMode(_cs, OUTPUT);
    wizchip_cs_deselect();

#if 0
    _spi.begin();
    _spi.setClockDivider(SPI_CLOCK_DIV4); // 4 MHz?
    _spi.setBitOrder(MSBFIRST);
    _spi.setDataMode(SPI_MODE0);
#endif

    wizchip_sw_reset();

    // Use the full 16Kb of RAM for Socket 0
    setSn_RXBUF_SIZE(16);
    setSn_TXBUF_SIZE(16);

    // Set our local MAC address
    setSHAR(_mac_address);

    // Open Socket 0 in MACRaw mode
    setSn_MR(Sn_MR_MACRAW);
    setSn_CR(Sn_CR_OPEN);
    if (getSn_SR() != SOCK_MACRAW) {
        // Failed to put socket 0 into MACRaw mode
        Serial.printf("DEBUG: Sn_SR : [%d]\r\n", getSn_SR());
        return false;
    }

    // Success
    return true;
}`,
what's wrong? Please help me! Thinks!!!

d-a-v, 2019-11-12:
You must initialize SPI before `ether.begin()`

mynameisbill2, 2019-11-14:
Thank you very much! The library worked correctly now!

closed 2019-11-14 by mynameisbill2