d-a-v/W5500lwIP #11: lwip dhcp non-blocking connection?

jbaudoux, 2019-11-14:
Does this provides non-blocking connection like in WiFi core with a callback when connection is successful?
ESP8266WiFi uses lwip/dhcp with callback: WiFi.onStationModeGotIP(onWifiConnect)
This allows to use an aync mqtt client like this one: https://github.com/marvinroger/async-mqtt-client
Note that PubSubClient makes blocking calls (timeout 20000)

In Ethernet, the dhcp is a blocking call with long timeout:
unsigned long timeout = 60000, unsigned long responseTimeout = 4000

d-a-v, 2019-11-14:
DHCP is not blocking but there is currently no callback.
This project is going soon to be obsoleted by [this pull request](https://github.com/esp8266/Arduino/pull/6680) on esp8266 arduino core.

The traditional `while (eth.status() != WL_CONNECTED)` or `while (!eth.connected())` is possible.

A callback can be added, let's follow this discussion on the PR.

jbaudoux, 2019-11-14:
Thanks for your answer. I think I got it. It's is not using Ethernet lib anymore, so the dhcp timeout issue there is not relevant anymore.
For async tcp call, I wonder how can the lib https://github.com/me-no-dev/ESPAsyncTCP know which interface to use?

d-a-v, 2019-11-14:
lwIP automatically selects the interface according to IP addresses (ie: peer address must match interface address - or default interface when no matching can occur - the ethernet interface can be selected to be the default one).

So this ethernet driver should work flawlessly and transparently with ESPAsyncTCP. It is in fact the very purpose of it.

jbaudoux, 2019-11-14:
Thanks for the clarification

closed 2019-11-14 by jbaudoux