d-a-v/W5500lwIP #14: UDP receive don't work.

nopnop2002, 2020-05-27:
Thank you for great work.

A TCP transmission, TCP reception, and UDP transmission work correctly, but UDP reception does not work.

This is my sketch:

	#include <SPI.h>
	#include <ESP8266WiFi.h>
	#include <WiFiUdp.h>
	#include <W5500lwIP.h>   // https://github.com/d-a-v/W5500lwIP

	#define CSPIN D2

	Wiznet5500lwIP eth(SPI, CSPIN);
	byte mac[] = {0x00, 0xAA, 0xBB, 0xCC, 0xDE, 0x02};

	// Your NTP Server
	#define NTP_SERVER     "pool.ntp.org"

	// Your local time zone
	#define TIME_ZONE      9            

	// local port to listen for UDP packets
	#define LOCAL_PORT      8888

	// NTP time stamp is in the first 48 bytes of the message
	#define NTP_PACKET_SIZE 48

	// NTP time stamp is in the first 48 bytes of the message
	byte packetBuffer[NTP_PACKET_SIZE];

	// Time of last packet transmission(ms)
	unsigned long lastSendPacketTime = 0;

	// A UDP instance to let us send and receive packets over UDP
	WiFiUDP udp;

	// send an NTP request to the time server at the given address
	void sendNTPpacket()
	{
	  Serial.println("\nsending NTP packet...");
	  // set all bytes in the buffer to 0
	  memset(packetBuffer, 0, NTP_PACKET_SIZE);
	  // Initialize values needed to form NTP request
	  // (see URL above for details on the packets)
	  packetBuffer[0] = 0b11100011;   // LI, Version, Mode
	  packetBuffer[1] = 0;     // Stratum, or type of clock
	  packetBuffer[2] = 6;     // Polling Interval
	  packetBuffer[3] = 0xEC;  // Peer Clock Precision
	  // 8 bytes of zero for Root Delay & Root Dispersion
	  packetBuffer[12]  = 49;
	  packetBuffer[13]  = 0x4E;
	  packetBuffer[14]  = 49;
	  packetBuffer[15]  = 52;

	  // all NTP fields have been given values, now
	  // you can send a packet requesting a timestamp:
	  udp.beginPacket(NTP_SERVER, 123); //NTP requests are to port 123
	  udp.write(packetBuffer, NTP_PACKET_SIZE);
	  udp.endPacket();
	}

	void setup()
	{
	  Serial.begin(115200);
	  Serial.println();
	  Serial.print(ESP.getFullVersion());
	  Serial.println();
	  
	  SPI.begin();
	  SPI.setBitOrder(MSBFIRST);
	  SPI.setDataMode(SPI_MODE0);
	  SPI.setFrequency(40000000);

	  eth.setDefault(); // use ethernet for default route
	  int present = eth.begin(mac);
	  if (!present) {
	    Serial.println("no ethernet hardware present");
	    while(1);
	  }
	  
	  Serial.print("connecting ethernet");
	  while (!eth.connected()) {
	    Serial.print(".");
	    delay(1000);
	  }
	  Serial.println();
	  Serial.print("ethernet ip address: ");
	  Serial.println(eth.localIP());
	  Serial.print("ethernet subnetMask: ");
	  Serial.println(eth.subnetMask());
	  Serial.print("ethernet gateway: ");
	  Serial.println(eth.gatewayIP());

	  Serial.println("Starting UDP");
	  udp.begin(LOCAL_PORT);
	}

	void loop()
	{
	  long now = millis();
	  if (now - lastSendPacketTime > 5000) { // 5 seconds passed
	    // Send time request to NTP server
	    sendNTPpacket();
	    lastSendPacketTime = now;
	  }

	  // wait to see if a reply is available
	  if ( udp.parsePacket() ) {
	    //Serial.print("packet received, length=");
	    //Serial.println(cb);
	    // We've received a packet, read the data from it
	    udp.read(packetBuffer, NTP_PACKET_SIZE); // read the packet into the buffer

	    //the timestamp starts at byte 40 of the received packet and is four bytes,
	    // or two words, long. First, esxtract the two words:
	    unsigned long highWord = word(packetBuffer[40], packetBuffer[41]);
	    unsigned long lowWord = word(packetBuffer[42], packetBuffer[43]);

	    // combine the four bytes (two words) into a long integer
	    // this is NTP time (seconds since Jan 1 1900):
	    unsigned long secsSince1900 = highWord << 16 | lowWord;
	    Serial.print("Seconds since Jan 1 1900 = " );
	    Serial.println(secsSince1900);

	    // now convert NTP time into everyday time:
	    // Unix time starts on Jan 1 1970. In seconds, that's 2208988800:
	    const unsigned long seventyYears = 2208988800UL;
	    // subtract seventy years:
	    unsigned long epoch = secsSince1900 - seventyYears; 
	    Serial.print("Unix time = ");
	    Serial.println(epoch);
	  
	  }
	}

This is Serial.print:

	SDK:2.2.2-dev(38a443e)/Core:2.7.1=20701000/lwIP:STABLE-2_1_2_RELEASE/glue:1.2-30-g92add50/BearSSL:5c771be
	connecting ethernet...
	ethernet ip address: 192.168.10.179
	ethernet subnetMask: 255.255.255.0
	ethernet gateway: 192.168.10.1
	Starting UDP

	sending NTP packet...

	sending NTP packet...

	sending NTP packet...

	sending NTP packet...

Do you have eany help.
Thank you.

d-a-v, 2020-05-27:
Can you try to use https://github.com/esp8266/Arduino/pull/6680 instead of this repository ?
It will be the one integrated in the esp8266 arduino core (and it is more advanced) (and it will be the one we will debug in case of issue)

nopnop2002, 2020-05-27:
Thank you for your quick response.
I will try.

d-a-v, 2020-06-02:
I [tried](https://github.com/d-a-v/W5500lwIP/issues/17#issuecomment-637731368) and UDP is running.

nopnop2002, 2020-06-04:
Will this issue be fixed in core version 3.0.0?

d-a-v, 2020-06-04:
Yes. You can already try gamma release (v0.0.3) at https://d-a-v.github.io .

nopnop2002, 2020-06-04:
Thank you again.