d-a-v/W5500lwIP #16: Example repo

nopnop2002, 2020-05-30:
Thank you for great work.

I made an example repo.

https://github.com/nopnop2002/esp8266_ethernet

If you read this, please close it.

Thank you again.

d-a-v, 2020-05-30:
You should really try this pull request https://github.com/esp8266/Arduino/pull/6680 which has a better integration and is an improvement of the driver you are using. If there is an issue with UDP, fix will be done in the pull request.

d-a-v, 2020-05-30:
There is an easier way to try the ethernet PR:
You can change the URL in your board manager to point to the one described there:
[Alpha, Beta and Gamma versions](https://d-a-v.github.io).
6680 is included in the version "gamma" 0.0.3. You need to "update" your version 2.7.0 or 2.7.1 to version 0.0.3 (which is version 2.7.1+ = 2.7.2beta plus the listed pull-requests). The ethernet driver will be included.
If there is a confirmed issue with UDP, then I'll have a look.

nopnop2002, 2020-05-30:
>There is an easier way to try the ethernet PR:

Thank you.

Regarding the NTP client, I found that configTzTime () works correctly.

https://github.com/nopnop2002/esp8266_ethernet/blob/master/configTzTime/configTzTime.ino